Psychokiller1888/Vertex-Delta-Firmware #3: stepps/mm | stepper

chuebsch, 2019-01-10:
Hi there thanks for the last quick fix, here comes an other minor issue: 
In the Velleman forum  we figured out recent the number of teeth, belt pitch etc.
so can you please update the steps per mm to:
```C++
// variables to calculate steps
#define XYZ_FULL_STEPS_PER_ROTATION 200
#define XYZ_MICROSTEPS 32
#define XYZ_BELT_PITCH 2.5
#define XYZ_PULLEY_TEETH 22

// delta speeds must be the same on xyz
#define DEFAULT_XYZ_STEPS_PER_UNIT ((XYZ_FULL_STEPS_PER_ROTATION) * (XYZ_MICROSTEPS) / double(XYZ_BELT_PITCH) / double(XYZ_PULLEY_TEETH))
#define DEFAULT_AXIS_STEPS_PER_UNIT   { DEFAULT_XYZ_STEPS_PER_UNIT, DEFAULT_XYZ_STEPS_PER_UNIT, DEFAULT_XYZ_STEPS_PER_UNIT,  154.895 }  
```
... I know the difference ins tiny but this way the values are not just fallen from the sky.
==========
next issue: 
I think a neat thing in recent Marlin is that on can specify the stepper drivers:
```C++ 
#define X_DRIVER_TYPE  DRV8825
#define Y_DRIVER_TYPE  DRV8825
#define Z_DRIVER_TYPE  DRV8825
#define E0_DRIVER_TYPE DRV8825
```

Psychokiller1888, 2019-01-10:
We don't need to have these values, they are hard set in the code

Psychokiller1888, 2019-01-10:
But I can set them just for fun :)

chuebsch, 2019-01-10:
thank you! :smile:

closed 2019-01-10 by chuebsch

Psychokiller1888, 2019-01-10:
Btw, these values are now tunable on printer

Psychokiller1888, 2019-01-10:
Especially the extruder set at 154.895 after measuring the real extrusion length