Psychokiller1888/Vertex-Delta-Firmware #5: Support for heated bed

Psychokiller1888, 2019-01-10:
The main board has a 14v output. Using 2 or three diodes it can easily be taken down to 12v for a 200w heated bed like 

https://www.ebay.com/itm/3D-Printer-Round-Silicone-Heated-Bed-12v-24v-heat-plates-Dia-180mm-260mm/192070208772

Thermistor value is 60

This setting has to be off by default and editable on the machine directly, for turning on and pin setting

chuebsch, 2019-01-13:
Never use the 14V output of the K8800 board for anything like a heated bed, or you will burn the bed or the power supply. You will need a 300W or more external power supply, an appropriate MOSFET and  a 4.7kOhm pull up resistor.
  
Follow this thread if you like:
https://forum.vellemanprojects.eu/t/building-a-heatbed-for-k8800/29826

I got the same silicone heat pad yesterday. I think thermistor value 75 is better suited for this as 60 is just generated by a script.

Psychokiller1888, 2019-01-14:
Hey, thanks for the info! Yeah, pretty much figured out I would have enough amps for 200w

About the thermistor, are you sure? 60 is the actual correct declared one?